mathiasvr/bluejay #2: requires higher idle than blheli_s

Quick-Flash, 2021-01-15:
With blheli_s I was able to happily run an idle value of 3. With this code I need to run an idle of 5.5 in order to keep all the props spinning in the air. Also the startup kick is just to much, need a way to dial that down.

mathiasvr, 2021-01-15:
Hi, yes the startup boost is currently not desirable and will be updated in the next release.
Interesting about the idle speed, I will try to look more into it and see if I can replicate it.

Which hex file did you use and did you have rpm telemetry enabled?

Quick-Flash, 2021-01-15:
I'll need to check which hex. I had 96khz flashed, not using rpm filtering.

mathiasvr, 2021-01-15:
In my testing, it seems the 48/96khz versions need a bit more startup boost to get spinning, maybe this means they also need more idle speed to maintain the same spin as the 24khz version?
However, I just tried out 96khz on a tinyhawk with 0802 motors, and I didn't really feel much difference between the 48khz version and had no problem with an idle value as low as 2.

Can you maybe try a 96khz version of BLHeli_M or JESC to compare?

Quick-Flash, 2021-01-15:
Sorry I was actually using blheli_m before. Not blheli_s. Maybe that's just a bl_m thing then. Either way thanks for the work your putting in here.

mathiasvr, 2021-01-16:
Okay no problem, I’ll try experimenting a bit with blheli_m to see if I can figure something out. 
Thanks for testing!

Quick-Flash, 2021-01-17:
i'll be testing 0.7 and let you know how things go.

Quick-Flash, 2021-01-25:
actually it seems that blheli_m and 96khz was acting a bit different... Went back to 24khz and things were acting a ton better.

mathiasvr, 2021-01-25:
Which motors are you using? I think it's only smaller motors that really benefit from 48/96khz.
I still need to do some more testing myself though.

howels, 2021-01-25:
FWIW I put 0.8 96KHz on a moblite7 with 0802 19000KV. No changes to defaults were needed to get motors spinning on BetaFlight 4.2.5. Probably I could lower idle but it's not too floaty with the defaults.

Quick-Flash, 2021-01-25:
> Which motors are you using? I think it's only smaller motors that really benefit from 48/96khz.
> I still need to do some more testing myself though.

i'm using 1204 motors. I like to test 96khz because a lot of users tend to like to push that up (even if its not always the best idea) So i like to be able to "push" my gear by going to 96khz just to see what i can do.

mathiasvr, 2021-01-25:
@howels Thanks it's always helpful for me to have more info on how things work out on different setups.

@Quick-Flash I read that increasing the pwm frequency also increases the effective dead time, so in that aspect you could say the motor is being "pushed" more when running 24khz.

It would be interesting to try out if increasing deadtime provides the same "benefits" as increased pwm, or if the increase in pwm frequency also has other effects.

closed 2021-02-11 by Quick-Flash

corrado-c, 2021-02-18:
> In my testing, it seems the 48/96khz versions need a bit more startup boost to get spinning, maybe this means they also need more idle speed to maintain the same spin as the 24khz version?
> However, [...]

Sorry if I re-open this issue,
but i am also testing with 0802 tinyhawk !!
idle throttle 8
96khz, timing 22 ° and then 15, rpm power first 9x then 11x
Very good first impression, very reactive but I think it's slightly underpowered at low rpm, I'm not very sure.
could you tell me the settings you recommend for these engines?
startup min and max, timing, demag, rpm power, dithering?

if there are other channels to write my considerations tell me :)
thanks for your fantastic work!

mathiasvr, 2021-02-18:
@corrado-c Hi there, I think this could be related to blheli_m having a nonlinear throttle curve which can sometimes help with this? I don't know if this is still/always the case but I've seen it discussed on rcgroups. Bluejay instead targets a linear throttle curve since it is easier to reason about on different setups.

Betaflight has a setting called `thrust_linear` to handle this issue of less thrust on low throttle.
I haven't actually used the setting myself but I just noticed that https://theuavtech.com/presets/ recommends trying `set thrust_linear = 25` when running pwm at 48/96khz. I'm very interested in hearing how it works for you if you try it out.

I mostly run the default settings of Bluejay but increased min. startup power to 1100 to compensate for a stubborn motor.
Dithering should generally be enabled when running 48/96khz since it increases the throttle resolution which gets lower with higher pwm frequency.

corrado-c, 2021-02-18:
yess :) I recently discovered the settings recommended by uavtech! I like them very much!
I'm already at thrust_linear 25. now i just tried 48khz, only one lipo, i think it is more powerful than 96 and i had more flight time than 96, but it seems strange and i should test more flights (the 6 lipo i have are worn and all different). in any case I like it very much!

mathiasvr, 2021-02-18:
Okay great, actually I just remembered that the tinyhawk has a dead time of 90 which is quite high. When running pwm at higher frequency this comes into play more often so 96khz on the tinyhawk could be impacting throttle control a lot - which is especially a problem at low rpm, so I think 48khz is much better here.

I hope to do more flight testing myself at some point to better understand which problems should be addressed in the esc firmware vs fc firmware ;)

mathiasvr, 2021-02-18:
Btw @Quick-Flash, do you know if Emuflight has something similar to `thrust_linear` or is considering thrust in some way?

Quick-Flash, 2021-02-18:
Right now emu doesnt have thrust linear. However it was recently added and will be in the next release. We are aso handling thrust linear differently than bf. We split it up so that yaw control does not use thrust linear only pitch and roll control do. That's because yaw is not effected by thrust.

howels, 2021-02-18:
> yess :) I recently discovered the settings recommended by uavtech! I like them very much!
> I'm already at thrust_linear 25. now i just tried 48khz, only one lipo, i think it is more powerful than 96 [...]

Try turning thrust_linear up to 35 or 40. Just experiment to see what works. If run between 35 and 40 on my 3" with JESC 48KHz and the same should apply to bluejay.

Shellodan, 2022-04-22:
I all I just flashed my meteor 65 with bluejay 96khz, now the props spin but it won't take off please help