mathiasvr/bluejay #4: blheli-configurator fork for Bluejay like JESC and others

howels, 2021-01-22:
Please consider forking blheli-configurator like others have done (https://github.com/Asizon/blheli-configurator/releases and https://github.com/jflight-public/jesc-configurator) because this makes it really easy to quickly update ESCs and test new releases.

thanks

mathiasvr, 2021-01-22:
@howels Yes, I'm trying :)

I basically had to do it since I have decided to move towards changing the internal layout of Bluejay which will make it incompatible with BLHeli configurators.

Though because of this, it might be some time before I can do a proper release.

Quick-Flash, 2021-01-22:
@mathiasvr you should try working with asizon on the configurator. He will probably be willing to help.

howels, 2021-01-23:
I think the blheli-m configurator re-used regular blheli-s options to convey the jazz maverick options. So if bluejay ends up being substantially different to blheli-s then I can see where there could be problems.

Quick-Flash, 2021-01-25:
> I think the blheli-m configurator re-used regular blheli-s options to convey the jazz maverick options. So if bluejay ends up being substantially different to blheli-s then I can see where there could be problems.

well at that point it would only be a problem if there isn't a bluejay configurator. If there is a bluejay configurator then this stuff shouldn't be a problem.

mathiasvr, 2021-01-25:
I will try to release a configurator that works on the old layout so we can have one a bit sooner.

mathiasvr, 2021-01-26:
I have now released a version here: https://github.com/mathiasvr/blheli-configurator/releases
Still very early work but it should make it easier to install - hopefully without issues.

howels, 2021-01-27:
> I have now released a version here: https://github.com/mathiasvr/blheli-configurator/releases
> Still very early work but it should make it easier to install - hopefully without issues.

Excellent. Connects to my board fine. Haven't tried to re-flash but looks promising :+1:

mathiasvr, 2021-02-11:
So we have a configurator now. I hope to be able to merge it back to blheli configurator when bluejay's settings layout becomes more stable.

closed 2021-02-11 by mathiasvr