mathiasvr/bluejay #5: Request for a new pinout

crteensy, 2021-01-26:
This one is apparently rather special. We'd like to use the following pinout:
- Ap: 1.1 (open drain active high)
- Ac: 1.0 (open drain active low)
- Bp: 1.2 (open drain active high)
- Bc: 1.3 (open drain active low)
- Cp: 1.5 (open drain active high)
- Cc: 1.4 (open drain active low)
- MA: 0.0
- MB: 0.2
- MC: 0.1
- CC: 0.3
- Tx: 0.4
- Rx: 0.5

We use NCP81253 gate drivers, which accept 3-state PWM input. This is generated with pull-ups and the mentioned open drain outputs:
[image]

When Ac is low, the PWM input is low. When Ac is high-Z, Ap can either pull PWM to 2V5 (Ap low) or leave it at 5V (Ap high-Z). This has sort of worked with a different pinout, but we were forced to re-route the ESC to this pinout because of space constraints.

crteensy, 2021-01-27:
This thread on rcgroups describes a slightly different previous attempt (with Ac and Ap swapped for low-side PWM): https://www.rcgroups.com/forums/showthread.php?3532875-BLHeli_S-ESC-iteration-2

It includes scope shots that show how the generated 3-state PWM input looks when it's produced with open drain outputs and the pull-ups.

mathiasvr, 2021-01-27:
Thanks, looks very informative.

Should anyone be interested in new pinouts I will just note that the current implementation only supports when the pairwise ordering of pwmfets and comfets is the same for all 3 phases (topological ordering). I will try to make it work without this requirement but it remains the recommended pattern for efficiency reasons.

@crteensy Would you by any chance happen to know why the O layout needs the comparator output inverted?

crteensy, 2021-01-27:
Why O needs the comparator output inverted (educated guess):

O: "like D, but low side PWM"
D: "com FETs inverted"

When the high side FET is the com FET, all bemf signs are flipped - including the sign of its gradient. I'm on mobile now so I can't draw diagrams to illustrate, but that is the reason and I was able to verify this with totally different hardware. 

I interpret the existence of O this way: O was created for a 1S ESC with a high side PFET and a low side NFET. The high side gate output must be active low, and the low side gate output must be active high to drive them both directly. The MCU is powered directly from the battery, which fully utilizes the absolute maximum rating for its supply voltage (4V2). This makes the overall schematic extremely simple (-> fishpepper).
I can only guess why low side pwm was chosen, but I think it's because that way the pwm output doesn't need to be inverted and ends up driving the (usually less lossy) NFET.

mathiasvr, 2021-01-27:
Thanks a lot that makes sense, I hadn't thought of that possibility.
So do you also need the inverted bemf output like O?

crteensy, 2021-01-27:
No we don't need inverted bemf. The ESC will do "classic" high side pwm with the pinout and gate polarity described at the top

mathiasvr, 2021-02-13:
[Here is a capture](https://github.com/mathiasvr/bluejay/files/5974884/DT90.sal.zip) of the pwm signals with a dead time of 90 (1.8us).
It can be opened with Logic 2 if you want to inspect it.