mathiasvr/bluejay #6: variable pwm

Quick-Flash, 2021-02-11:
Some sort of way to make the pwm dynamic. This has been implemented in am32 and is called variable pwm. Something worth looking into if it isn't to difficult, but I do understand how difficult a lot of this coding must be.

mathiasvr, 2021-02-11:
I think this will be difficult on the BBs if it's even possible. I don't know much about variable pwm so it will probably be long before I can take a look at this. Let's keep tracking it here, maybe someone has more information.

senbax, 2022-02-20:
AM32 has support for variable PWM and is open-source. Maybe this gives some hints on how to implement variable PWM: https://github.com/AlkaMotors/AM32-MultiRotor-ESC-firmware

closed 2025-01-20 by Quick-Flash