mathiasvr/bluejay #7: PWM frequency selection without reflashing

lmnn, 2021-02-11:
Hi,
would be nice to select PWM frequency in Configurator without reflashing the firmware.
Like in blheli_m configurator:
[image]

mathiasvr, 2021-02-11:
Hi, yes it was mentioned at one point.

Currently, PWM is updated in a way (like BLHeli_S) that makes this a bit complicated.
I will have to examine this method further to figure out what is the easiest way to support this, but it's not the main priority right now so might take a while.

stylesuxx, 2022-01-30:
This can be done with the latest [pre-release](https://github.com/mathiasvr/bluejay/releases/tag/test-melody-pwm)