mathiasvr/bluejay #10: Comparison between bluejay and BLHELI_M

stylesuxx, 2021-02-17:
Hey, I was wondering if it would be possible to get a short summary in the readme what the difference between bluejay and BLHELI_m are.

I mean, apart from this one being properly source controlled ;-)

Thanks for the work anyway, will keep an eye on this project.

mathiasvr, 2021-02-17:
Hi, sure I’ll see what I can do.

Most significantly blheli_m uses a different method of commutation which can be configured using the settings ff/ce iirc. 

Bluejay does not support these settings and uses mostly the old blheli_s logic to drive the motors.

However, I haven’t really used blheli_m much and don’t know how it works in detail, and since Jazz has not publish his latest source code I can’t really look into it.

Maybe someone with more experience / knowledge can help out?

stylesuxx, 2021-02-17:
Oh, I was not aware that the source was not public. I thought it was just buried somewhere in the rcgroups thread...

Maybe this is a bit out of place here - but what do you mean by "different method of communication"? Or to be more specific, communication between which components? Is there a slack(or similar) one could join for same quick Q&A?

Really interested in a "proper" OSS telemetry capable ESC firmware.

mathiasvr, 2021-02-17:
> Oh, I was not aware that the source for BLHELI_M was not public. I thought it was just buried somewhere in the rcgroups thread...

I suppose I could be mistaken, but the most recent source code I see on the github repo is version 16.8_rc3. I think we may assume this is mostly version 16.8, but the latest version 16.9 is not there.

> Maybe this is a bit out of place here - but what do you mean by "different method of communication"? Or to be more specific, communication between which components? Is there a slack(or similar) one could join for same quick Q&A?

I'm referring to motor *commutation* (not communication), which affects how motors are driven. BLHeli_M introduced new methods in an attempt to improve efficiency among other things. It hasn't worked well for all setups but I believe that BLHeli_M is now flexible enough that it can be properly tuned using the settings to suit different types of ESCs, motors, etc. However, it can be a bit complicated to understand and get these settings right. Async PWM is what should be most similar to Bluejay but as mentioned I don't really know much about this.

I don't actually know where most people discuss stuff like blheli_m? 😅 
I think it would be a good idea to at least set up some channel for Bluejay to just have something. I will look into that.

stylesuxx, 2021-02-17:
> I suppose I could be mistaken, but the most recent source code I see on the github repo is version 16.8_rc3. I think we may assume this is mostly version 16.8, but the latest version 16.9 is not there.

Yes, you probably are right with that assumption.
 
> I'm referring to motor _commutation_ (not communication), which affects how motors are driven. BLHeli_M introduced new methods in an attempt to improve efficiency among other things. It hasn't [...]

Sorry, I obviously was not reading carefully enough - makes sense now.
But yeah - it would definitely be interesting to get all active BLHELI_S devs into a room to see what their firmware does and why they chose the way they went. Jazz and Thorsten, the JESC dev.
 
> I don't actually know where most people discuss stuff like blheli_m aside from rcgroups?
> I think it would be a good idea to at least set up some channel for Bluejay to just have something. I will look into that.

Discord seems to be quite popular with the FPV dev community right now - at least for chatting. I think real issues and discussion should take place here anyway, but a chat is good for quickly asking some questions...

On a different note: How stable would you consider your firmware to be? Is it ready for prime time yet? On what kind of setups has it been tested? I have a small blog (mainly whoops and micros) and might write a couple of words, show how to install and configure. But I don't want to blast you (and me) with loads of support requests if it's not there yet.

mathiasvr, 2021-02-17:
> Discord seems to be quite popular with the FPV dev community right now - at least for chatting. I think real issues and discussion should take place here anyway, but a chat is good for quickly asking some questions...

Yes, I think Discord would be a good option.

> On a different note: How stable would you consider your firmware to be? Is it ready for prime time yet? On what kind of setups has it been tested? I have a small blog (mainly whoops and micros) and [...]

Thanks a lot that's very nice. I do not really consider it stable yet and I have not personally tested it on a lot of setups. It's only really been used by a few people up to this point. I have quite a few changes planned as well so maybe it's still a bit early for the post? That being said, I do rely on the community to test it on different setups and try hard to avoid issues in every release. As long as people take precautions and spend time testing for any potential issues I think it should be fine - so not really prime time 😉 

The actual first stable release will be Bluejay 1.0 and it might be quite some time before we get there.

stylesuxx, 2021-02-17:
OK, I will give it a spin on a couple of my whoops and see how I like it, at least I could provide some compatibility data. I would also be interested in the changes you have planned, maybe that would also be a good section for the Readme to see in which direction the project is heading.

mathiasvr, 2021-02-17:
Thanks, I might see if I can sketch something out but most of it is optimizations, bugfixes, and endless refactoring. From a user perspective, I'm not sure much will change aside from the addition of beep melodies.