mathiasvr/bluejay #12: Daily/Nightly builds

stylesuxx, 2021-02-22:
Now that we have a pipeline for building, I would highly suggest a build server creating daily/nightly builds. I can chip in space and the actual build pipeline. I would build the main branch and keep a 7 day backlog of those builds.

If it wasn't for the crappy licensing situation I think integration tests via travis would be pretty cool - we could run the docker build pipeline via travis and would know if a commit breaks our builds.

DusKing1, 2021-03-10:
Travis seems does not provide free plan for general users anymore. How about github action or Azure pipeline?

mathiasvr, 2021-03-10:
@DusKing1 Sounds good to me.

It could also be nice to make it automatically upload/prepare a release when a commit is tagged.

gretel, 2021-03-25:
`github actions` can also store [artifacts](https://docs.github.com/en/actions/guides/storing-workflow-data-as-artifacts). not exactly related to time of days but suitable i think 😏