mathiasvr/bluejay #14: v0.10 does not go to full speed

KingKone, 2021-02-26:
Hi, first of all, thx for your great work!

Just tested v0.10 and the reported RPM stops around 3700-4000, it is spinning to slow by the sound of it.

After I have noticed that, I checked with v0.9 and reported RPM is around 32000 (empty 5S 1750KV), sounds right again.

ESC Config is on defaults.
 - Bluejay version: 0.10 (24 & 48 kHz)
 - ESC variant:  C-H-40 (Mamba F405 MK2 Stack)
 - Bidirectional DShot: On
 - FC firmware: BF 4.2

If you need me to test something, I am up for testing.

mathiasvr, 2021-02-26:
Hi, thanks for the report!

That sounds strange, having the rpm protection setting very low could cause this, are you sure it was at default of 9x?
If that's the case I will have to examine the C layout more closely.

mathiasvr, 2021-03-09:
@KingKone Can you confirm this happens with the default settings?

koen352003, 2021-04-18:
I'm using the same mamba f405 mk2 stack, and for me 0.12 (test1) has been working and flying perfectly fine on defaults.

mathiasvr, 2021-04-18:
@koen352003 Thank you for the report!

I don't know what could have caused this issue other than the mentioned setting, so it's strange using v0.9 would work - maybe the settings were not the same for some reason.
Anyway, I think we can close this for now then.

closed 2021-04-18 by mathiasvr

ericogr, 2021-04-19:
> I'm using the same mamba f405 mk2 stack, and for me 0.12 (test1) has been working and flying perfectly fine on defaults.

Same here. Today, experimented `4x1 Mamba F40HV ESC` with `C-H-40, 0.12, 48 kHz, Bluejay (BETA)` with 6S battery with success!
Params:
```
startups: 1010/1028
timing: mediumHigh
demag: high
rampup: 5x
```