mathiasvr/bluejay #17: Small delay on motor spin up when in turtle mode

FabianIsensee, 2021-03-14:
Hi there,
first and foremost: THANK YOU so much for providing this fantastic ESC firmware. It must be a great deal of work to program and maintain this. I don't know why there are only so few people using BlueJay thus far - it perfectly slides in the huge gap left between pay-to-use JESC and JazzMav (I don't have to say anything about that do I?). BlueJay ticks all the boxes: RPM telemetry, free to use and a maintainer that is able to communicate with and listen to their users :-)
I have installed V 0.10 this morning on a Mobula 6 whoop board. Everything works great out of the box (default settings, RPM enabled) and it just works.
There is just one thing I noticed: when I am using turtle mode, the motors do not respond immediately to stick inputs. There is a small delay that was not there with JESC/BLheli_s. Is that delay intentional? It makes it more difficult to time short motor pulses to turn the quad back upright and makes one feel less connected. If that is indeed intentional - what is the rationale behind it? I can hardly imagine that this is safer for the motors (I tend to overshoot) and it also feels less nice to use.
Best,
Fabian

FabianIsensee, 2021-03-14:
I just tested it vs a second, identical quad that is still running JESC. The perceived difference  between the two in LOS is way less than in the goggles. Maybe the motors don't spin up as fast in BlueJay (start is not delayed much but maybe thrust is?)? It could also be related to startup power, I will investigate further

mathiasvr, 2021-03-14:
Thanks a lot! 😊 

There is no intent to add a startup delay or decrease thrust in turtle mode so it could be a bug.
What pwm frequency and dead time are you using?
When running 48/96khz it seems more power is needed to spin up the motors, however turtle mode is not supposed to need even further power.

FabianIsensee, 2021-03-14:
Hi there. I can test some more things tomorrow. In the meantime: it's 48kHz and deadtime is 5 (O-H-5).
Best,
Fabian

FabianIsensee, 2021-03-21:
Hi, sorry for keeping you waiting so long. I had a busy week. Finally today I found some time to do thorough testing. I am lucky to have two identical quads which makes it easier for me. Both use 0803 19000kv motors, mobula6 AIO (Crazybee Lite, Frsky, O-H-5). One has bluejay 0.10, the other JESC 2.3. I made sure to match the ESC settings as much as I could (Startup 1.0 on JESC, 11x on bluejay. Demag comp. low. Motor timing medium). 
The delay with which the motors start to spin is about the same. Once the motors have spinned up, the responsiveness between the two is also the same. The only (very noticeable!) difference is the speed with which the motors spin up from standstill. bluejay is much slower in that regard than JESC causing turtle mode to feel laggy and unresponsive. I understand that this may be a safety feature (you seem to use RPM data to spin up the motors safely) but in the case of turtle mode at least this is not necessarily desired because the pilot may overshoot their commands (and thus maybe fry something) because they are trying to fight the lag.

mathiasvr, 2021-03-21:
Thanks, I will try to look into it as well.

BLHeli_S/JESC also has rpm protection to avoid excessive power, did you have that enabled?
Overshooting or high acceleration should preferably not be able to fry the ESC because of startup/rpm protection.

mathiasvr, 2021-03-21:
I tried comparing startup to full throttle between bluejay, jesc and blheli_s but don't experience much difference.
Can you post a screenshot of your configurator settings?

FabianIsensee, 2021-03-28:
[image]

[image]

Best,
Fabian

FabianIsensee, 2021-03-28:
After posting the screenshots I noticed that I forgot to set min and max startup power in bluejay. After doing that the effect is much less pronounces. It will probably disappear by increasing them even more. I am not sure however whether that could result in damage to the motors or escs. The values are now at 1050/1200

mathiasvr, 2021-03-28:
With the updated startup values it should be mostly equivalent to JESC so it’s interesting that it’s still not quite there.

It would be great if you could try reducing rpm rampup to 9x or lower to see if the problem is only related to max startup power?

You can also reduce min startup power if the motors have no problems starting at low throttle input.

The reason for all these settings was to be able to increase startup power without also increasing rampup - which needs to be lower on some ESCs.

closed 2021-08-06 by FabianIsensee