mathiasvr/bluejay #19: Play melodies with the dshot beacon

mathiasvr, 2021-04-04:
We can now change the startup melody, but we also want to play melodies with the beacon signal.

Quick-Flash, 2021-04-05:
Yes that would be fun to allow the dshot beacons to just turn on esc music rather than one tone of beeps. Would be fun to be searching for your drone only to have it rick rolling you.

koen352003, 2021-04-06:
> We can now change the startup melody, but we also want to play melodies with the beacon signal.

I think I might have missed something, are you actually saying that the custom startup melody is already working? Didn't notice it being in the changelog, but if it's true than that's really quite amazing! I really like the work you and the others are doing around here, it's wonderfull to see this amount of support for the blheli_s esc's!

mathiasvr, 2021-04-06:
@koen352003 Yes might be a bit overstated.

Version [0.12](https://github.com/mathiasvr/bluejay/releases/tag/v0.12-test) is currently in a test phase. You can try it with the newest version of bluejay configurator and soon esc configurator too. It isn't perfect yet but can play some simple rtttl melodies.

howels, 2021-10-02:
Is it possible to have a beacon melody in addition to the startup melody? I urgently need to be able to rick-roll my fellow flyers.