mathiasvr/bluejay #24: Direction is always normal

GRIzmak, 2021-05-05:
**Describe the bug**
I'm not able to set motor direction. Whatever I choose (normal or reverse) I have a normal direction.

**Expected behavior**
Been able to choose a motor direction. 

**Configuration:**
 - Bluejay version: 0.12.1
 - ESC variant: O-H-5
 - Bidirectional DShot: On
 - FC firmware: Betaflight 4.2.8 
 - FC: Happy model Crazybee F4 Lite
 - board_name: CRAZYBEEF4FS

mathiasvr, 2021-05-05:
Thanks for reporting this!

Version `0.12.1` was backported from newer code and released early because it contains an important fix.
However, looks like the backporting caused a typo that removed the part that decodes settings such as direction 🤦 

I have fixed it and will release it as `0.12.2` ASAP.

mathiasvr, 2021-05-06:
Fixed by https://github.com/mathiasvr/bluejay/commit/e686e05b8d67c9682c8302233091213fb0b56e55 - released in [v0.12.2](https://github.com/mathiasvr/bluejay/releases/tag/v0.12.2).

Thanks again for your report! 👍

closed 2021-05-06 by mathiasvr

GRIzmak, 2021-05-06:
Thanks for a fast fix. It's working now. I love the melodies on 8bit ESC. Now my 1S toothpicks are even cooler.