mathiasvr/bluejay #26: Elevated bidir DShot error rate

harry1453, 2021-05-12:
**Describe the bug**
Non-zero (5-10%) DShot telemetry error rate in betaflight configurator, accompanied by occasional "clicking" noise coming from motor

**Expected behavior**
No telemetry errors, motor runs smoothly

**Configuration:**
 - Bluejay version: 0.12.2
 - ESC variant: G-L-30
 - Bidirectional DShot: On
 - FC firmware: Betaflight 4.2.9

harry1453, 2021-05-12:
I've just realized this may be because I am using DShot600. JESC doesn't support DShot600 on L ESCs. Is this the case with BlueJay as well?

howels, 2021-05-13:
The older L hardware doesn't support dshot600 with telemetry. Not enough CPU power to process telemetry above dshot300 speeds

mathiasvr, 2021-05-13:
Yes, as @howels mentioned the mcu is too slow to handle it. I would recommend using dshot150. I’m not sure how dshot300 performs but I’m working on some optimizations that may improve it.

I haven’t tested an L ESC so the telemetry signal has not yet been fine-tuned and that’s also why I didn’t disable support for dshot600. We should probably remove dshot600 on L ESCs in next update to avoid this behavior?

howels, 2021-05-13:
> Yes, as @howels mentioned the mcu is too slow to handle it. I would recommend using dshot150. I’m not sure how dshot300 performs but I’m working on some optimizations that may improve it.
>
> I haven’t [...]

Agree, safer to disable dshot600. Better if people can't take off than if they can fly but get unexpected accidents.

harry1453, 2021-05-16:
Okay, that makes sense. I suspect the clicking noise I heard was the MCU missing or mistiming commutations due to being too slow. Could probably cause desyncs if this is the case. I think the safer option is to leave it disabled. I'll give DShot300 a go on L hardware.

mathiasvr, 2021-07-11:
DShot 600 support for L ESCs has now been removed in Bluejay v0.14. I added some info no the wiki.

closed 2021-07-11 by mathiasvr

Quick-Flash, 2021-07-11:
> DShot 600 support for L ESCs has now been removed in Bluejay v0.14. I added some info on the wiki.

Can you re-enable this as long as your not using dshot telemetry? So bidir dshot will only work at dshot 300, but not dshot 600. But dshot 600 non bidir will work.

mathiasvr, 2021-07-11:
@Quick-Flash Ahh I don't have an L ESC so missed that performance might be good enough without rpm telemetry? Do you or anyone else run dshot 600 on L without issues? I'm not actually sure if it's mainly a performance issue or telemetry timing issue.