mathiasvr/bluejay #27: ESC lost after upgrade

AndreaCCIE, 2021-05-20:
**Describe the bug**
I lost one ESC after the upgrade from 0.11 to 0.13

[image]

**Expected behavior**
All 4 sec should be in the list

**Configuration:**
 - Bluejay version: 1.3.6
 - ESC variant: S-H-50
 - Bidirectional DShot: 300
 - FC firmware: Betaflight 4.3

mathiasvr, 2021-05-20:
Try waiting some time before pressing read setup, does the ESC still not show up?
Could you save debug logs and post them here?

AndreaCCIE, 2021-05-20:
No difference. I flew a pack (it's a tiny whoop), went for the upgrade and got the problem.

[Uploading Log.txt…]()

mathiasvr, 2021-05-20:
Looks like it failed to flash, but seems it should still show up. Can you try if esc-configurator.com is able to read it?

AndreaCCIE, 2021-05-20:
Yes, it does see the 4 escs but it doesn't spin (number 3)

[image]

AndreaCCIE, 2021-05-20:
Actually, it doesn't show number 3

[image]

mathiasvr, 2021-05-21:
Okay, looks like you have to flash using C2 interface to recover the esc unfortunately. This is not straightforward and you will need an Arduino or similar.

I'm not sure why this happened, but maybe there is some state bug in blheli/bluejay configurator - seems to be quite rare though.

mathiasvr, 2021-06-07:
There was a bug in blheli/bluejay configurator that could brick the ESC if the `Flash` button was pressed more than once.
I have fixed it in the latest version. Not sure if this is what happened in your case, but I hope we can avoid these problems in the future.

closed 2021-06-07 by mathiasvr