mathiasvr/bluejay #28: Support for single rotor helicopters

pkaig, 2021-06-01:
Hi. I'm using Bluejay for configuring the main and tail rotors (betaflight passthrough) on my helicopter running Heliflight-3D. I would like to be able to write a different setup to each motor and disable damped light mode for the main motor. Can these options be added please?

mathiasvr, 2021-06-07:
Hi there!

I'll try to see if I can add an option to disable "damped light" soon.
Let me know if you want some "test" files in the meantime.

For multirotors it probably makes the most sense to have this option affect all ESCs, so I'm not sure it should be a setting for each individual ESC - maybe there is a better way to support helicopters? Maybe there should be a mode in the configurator to allow having different settings individual ESCs. Are there any settings that it makes sense to share between motors in a helicopter?

pkaig, 2021-06-08:
Thank you very much Mathius. That would be great. Yes I would love to try out your test files please. I did read some comments in the project file that suggested this was being worked on.

The reason for the ability to download separately is because the main and tail motors are different so that can operate with vastly different settings such as timing and starting torques etc (the main can often be geared too). Damped light works well on the tail; however, on the main it causes the tail to whip around when you disarm or throttle hold. If this happens in flight (become disoriented) the tail rotation generally causes extra damage.

mathiasvr, 2021-06-08:
Ah okay, that makes sense!

@stylesuxx How complicated do you think it would be to have a mode/option where all settings are set individually in the configurator?

mathiasvr, 2021-06-08:
Here are some test hex files without braking/damping: [v0.13_nobrake.zip](https://github.com/mathiasvr/bluejay/files/6618061/v0.13_nobrake.zip).
I only compiled the most common ones, so let me know if you need one that's missing.

stylesuxx, 2021-06-08:
Technically I could add an option to treat every setting as an individual setting in the configurator.
But I am wondering how much of a fringe case this is - if it is just @pkaig who would be using this feature I am not sure if it is worth the hassle, or if it is easier for him to just de-solder the ESC signal wires and flash separately like that.

@pkaig I don't know the helicopter community, is this something more people would be interested in?

pkaig, 2021-06-08:
@mathiasvr. Thank you very much for the files. I will give them a go.

@stylesuxx. The added difficulty I have is I am using an AIO board. My hack to get around this is to download the tail config to both, then open in betaflight CLI and remove the resource for the tail motor. Reconnect Bluejay and flash the main motor config, reconnect betaflight and add the tail motor resource. It does work but isn't ideal. I guess looking at the current Bluejay Configurator there is an option to download the firmware individually on each ESC with a (flash all) at the bottom to do all. But there is only a single write setup. I'm not sure why you would want to load different firmwares and not setups. My hope was that it wouldn't be a big task to add a button on each ESC tab. 

As for how many would use it... I would 😃. I'm not sure how many others. I would assume a few. Previous blheli versions had this ability and BLheli_32 does as well. Not sure why it was removed from _S. 

Thank you both for your efforts on this project. It's a great bit of software!

stylesuxx, 2021-06-09:
@pkaig I see - an AIO makes it more complicated, that is true. I added your request as a feature request in the [esc-configurator repo](https://github.com/stylesuxx/esc-configurator/issues/113). I am not sure yet which version I will implement. If you have any preferences or other ideas, feel free to discuss them there.

Unfortunately this will be a bigger change so I can not promise a quick implementation.

pkaig, 2021-06-09:
@stylesuxx mate that sounds fantastic. I really appreciate you even considering the addition. I understand completely. Good things take time 👍

stylesuxx, 2021-06-10:
@pkaig could you have a look here => https://50e6a41ca189.ngrok.io/ in the settings you can now select "Disable common settings" - this will make the configurator show all settings for every ESC, let me know if this is what you were looking for.

pkaig, 2021-06-10:
@stylesuxx that's fantastic. Just connected and changed the starting parameters for the tail motor only. It seems to work great. That makes it so easy thank you very much.

stylesuxx, 2021-06-10:
Cool, thanks for giving it a spin. This means I can add it to https://esc-configurator.com

pkaig, 2021-06-11:
@stylesuxx great news. I will give this a go on my other helis when I get home. Again thank you.

@mathiasvr . Hi Mathias. I gave the noBrake firmwares a go but it still seems to be doing the damped-light. I took a video https://youtu.be/ZyLMujPY0YQ. Just reading a bit of your code in the 'push' section and I'm not sure if the damped-light is still enabled by default or if I am doing something incorrectly. I flashed the 92kHz firmware (if that matters). Cheers, Phil

mathiasvr, 2021-06-11:
@pkaig Braking/damping should be disabled by default (and not yet changeable). Did you also disable brake on stop?

pkaig, 2021-06-11:
@mathiasvr..... Ummmm. It was a Iid10t error.... I'm very much embarrassed to say I flashed the noBrake to the TAIL 🙃. That's what I get for rushing to give it a go before I head off to work. 

Anyway it works fantastic. I'm very excited to get this working on all my heli's now. I appreciate your efforts.

mathiasvr, 2021-06-11:
@pkaig Ah I see, great to hear! 😄

pkaig, 2021-08-29:
Hi Mathias, Thank you very much. I will give it a go.

pkaig, 2021-09-11:
Sorry for my very late reply Mathius. 

The damped light does make a big difference and with this enabled the heli does still spin rapidly. I have taken some videos. 
Your original v0.13 no damped light/no brake on stop: https://youtu.be/EsAWMqxvM7M    (this is great BTW)
The new no brake on stop V0.14: https://youtu.be/drHNzsOl2N0

The new version with damped light still enable spins much faster on throttle hold.

pkaig, 2021-09-11:
Sorry. Don't know what I clicked to close it.

mathiasvr, 2021-09-11:
Thank you @pkaig, the videos are very helpful! I see the stop still has too much braking for the helicopter landing. I'm surprised it has this much effect since the braking is very brief, but that's good to know.

Out of interest how is flight performance of the helicopter when flying damped light? I think there should be a noticeable difference when varying throttle? Do you think this is something that could be useful for helicopters or is no damping generally preferred?

I'm curious if I should look into removing the braking effect when stopping completely to allow helicopters to still fly damped light without the landing problem, however not sure if this is useful.

In any case I plan to include a braking strength setting that allows disabling damped light in v0.15.

pkaig, 2021-09-11:
I haven't flown it yet with your v0.14 but I will in the morning. Both of my helis running Bluejay have been broken and waiting for parts. One is ready to go now. 

Just thought I would Post a comparison between the standard V0.13 and your V0.13 nobrake (no damped light). It's really a huge difference.

V0.13: https://youtu.be/ahWzhT5Bn28
No Brake https://youtu.be/a2GtaOZoZYo

pkaig, 2021-09-11:
The no damped light only on stop may be worthwhile. The heli is running a fixed throttle with a Govenor to maintain headspeed regardless of pitch. I'm guessing damped light could help with rapid downward transitions that unload the head. 

Happy to do some testing.

mathiasvr, 2022-01-13:
Now v0.15 is finally released.
Setting the new setting  `Max Braking Strength` to 0 will effectively disable complementary PWM (damped light).

closed 2022-01-13 by mathiasvr

pkaig, 2022-01-18:
Thank you both @mathiasvr and @stylesuxx. v0.15 is now running on my E180 helicopter with GHF420 AIO board (and shortly to all my others). Tail motor has Max_Braking_Strength as default and the Main motor is set to 0. It works fantastic. Really great stuff.