mathiasvr/bluejay #32: The O_L_ layouts do not work

mathiasvr, 2021-08-10:
**Describe the bug**

The `O_L_` layouts have a comparator bug that causes incorrect back-emf readings making it impossible to drive the motor.

**Configuration:**

 - Bluejay version: 0.11+
 - ESC variant: O_L_X_X

howels, 2021-12-21:
Just discovered this. Doh.

howels, 2022-01-04:
Any likelihood of a fix for O_L @mathiasvr or is this a lower-priority item?

mathiasvr, 2022-01-04:
@howels It's fixed in a test release of `0.15` which you can already try flashing, but might be some time before esc configurator supports settings for that version.

swilkens, 2022-01-04:
Actually, esc-configurator should support 0.15 very soon;

https://github.com/stylesuxx/esc-configurator/issues/163

howels, 2022-01-04:
> @howels It's fixed in a test release of `0.15` which you can already try flashing, but might be some time before esc configurator supports settings for that version.

> @howels It's fixed in a test release of `0.15` which you can already try flashing, but might be some time before esc configurator supports settings for that version.

There are no O_L layouts in the 0.15-test release, can you add those if it's working please?

mathiasvr, 2022-01-04:
@howels Ah sorry, forgot I only added a subset of files 🤦 I'm adding the O_L as well.

manikbh, 2022-01-12:
Not sure if it is related, but when I flashed my old racerstar blheli_s 30A ESCs (G_L_30 from the read config) the melody changed but the motors were not running at all, except one, but it had failed to flash. Parameters could not be set (whatever I chose in the dropdown list was replaced by a blank value). I flashed back to blheli_s and it worked again.

mathiasvr, 2022-01-13:
Fixed in v0.15 which is now released.

mathiasvr, 2022-01-13:
@manikbh This issue only affects the O_L layouts, so problems with G_L_30 should be something else.

closed 2022-01-13 by mathiasvr