mathiasvr/bluejay #34: Extend DShot ESC Telemetry

mathiasvr, 2021-08-16:
It has been suggested to extend ESC telemetry to provide information such as missed zero crossings and demag events.

Currently Bluejay only supports rpm (e-period) telemetry according to the specification of bidirectional dshot.
DShot also allows telemetry on a separate wire which is currently only supported by 32-bit ESC firmware.

It needs to be decided whether Bluejay should add support for dshot (separate wire) telemetry and/or if bidirectional dshot (single wire) telemetry should be extended to somehow support different types of information than RPM.

Quick-Flash, 2022-04-14:
Fettec S2M protocol may be what we need. If it were to be supported on bl32 or bl_s through bluejay then I would believe we would see it become supported in bf.

damosvil, 2022-06-21:
There is no need to extend telemetry on a separate wire. DHOT telemetry coding repeats almost 50% coded values, so there exist a lot of codification values 7 ranges of 256 values to codify them making the pertinent modification in Betaflight.
Please check the explanation at https://github.com/betaflight/betaflight/issues/11677