mathiasvr/bluejay #35: Configurator Motor control screen/panel difficult to find or disappears

cybergap, 2021-09-01:
Screen for controlling and testing motors not always findable? Sometimes it is there, and sometimes not.

Seems like a bug in the configurator, else there is no obvious way to pull up the panel using menu or button. If not a bug, this is a feature request.

**Configuration:**

MacOS and Chrome based configurator
 - Bluejay version: [e.g. 0.6]
 - ESC variant: [e.g. A_H_5_24]
 - DShot bitrate: [150/300/600]
 - Bidirectional DShot: [On/Off/Either]
 - FC firmware: [e.g. Betaflight 4.2.6]

If your issue is related to *flashing*, please press `Save Debug Log` in the configurator and post the log.

mathiasvr, 2021-09-02:
Hi there,

This is the repo for the Bluejay ESC firmware, I think you are talking [ESC Configurator](https://github.com/stylesuxx/esc-configurator) since it has a motor test page, or did you mean another configurator?

cybergap, 2021-09-02:
Hey Mathias!

Yes, I couldn’t find a way to get to the test page. Didn’t know if I was just missing it or doing something wrong, but I checked again last night and couldn’t find it on the stand alone Mac version. 

Thanks for all your work on Blue Jay. It seems to be working great on my HX115 LR, after some fiddling with startup power and idle settings. 

Cheers

650-245-9354

mathiasvr, 2021-09-02:
Ah I see, the standalone bluejay configurator doesn't have a motor test feature, it's only on www.esc-configurator.com - you can of course also use betaflight configurator etc.

Thank you! And yes the smaller motor usually needs higher startup power, I've considered increasing the defaults. 
The HX115 looks very cool! I hope to get or build something very similar at some point 😉

cybergap, 2021-09-02:
Clears it up! Thanks 

Yes, the HX is actually amazing. Great little flier with 65mm blades and Nick Burns’ tune. Or, 1202.5 motors and 3” blades, and Blue Jay. One of my favorites now. So quiet. 

650-245-9354