mathiasvr/bluejay #36: Z-H-30 target does not work for Betafpv 1S 12A FC

longxk, 2021-09-25:
**Describe the bug**

Z-H-30 target does not work for Betafpv 1S 12A FC.
Tried 24khz and 48khz, motors dont spin.
Factory blheli_s 16.7 firmware works fine.

**Configuration:**

 - Bluejay version: [0.14]
 - ESC variant: [Z_H_30_48]
 - DShot bitrate: [300]
 - Bidirectional DShot: [Either]
 - FC firmware: [Betaflight 4.2.9]

longxk, 2021-09-26:
sorry. I just found out that I need to increase Min. Startup Power to at least 1100.
but what about Max. Startup Power and RPM Power? what values should I use?

mathiasvr, 2021-09-26:
Hi there, you should be fine with defaults, but based on min startup you can probably increase max startup to 1200 and 11x for rampup.

longxk, 2021-09-28:
thanks for your advice. set min to 1110, max to 1200 and rampup to 11x. but the motors wont spin properly on low throttle, just some twitches. is that normal?

mathiasvr, 2021-09-28:
You can increase dshot idle in betaflight to avoid low throttle problems.
48khz usually needs higher startup power / idle, but 24khz should be similar to blhelis. I will try looking into it and see if I can reproduce this issue.

JAL513, 2021-09-28:
I get this problem on 96khz as well. Not with all motors and they will start up when I turn up protection and ramp up power. Any other settings to improve this would be greatly appreciated. But will still happen intermittently.

JAL

mathiasvr, 2021-09-28:
I just tried this target with a 0802 motor and do not see any noticeable difference between BLHeli_S 16.7 and Bluejay 0.14 (24kHz), both starting at around 1060 throttle at default settings.

With 48kHz PWM the max startup power setting needs be increased (to e.g. 1200). This is normal with higher PWM frequencies at low throttle. The DShot idle value should also be increased to keep the motor running.
With 96kHz power must be increased even further.

The needed settings depends on the motors, ESCs, battery power etc. It is best to use new clean motors to tune the settings since dirty motors will need more power which risks of burning them.

You can do a motor test in betaflight or esc configurator to determine the throttle value that is needed to keep the motors spinning reliably.  You probably need to raise throttle to a higher value first to start the motors and then back down to find how low you can go without them stopping. You want DShot idle to be a bit more than the lowest value that keeps them spinning, e.g. for me 1060 was min. needed to start them which corresponds to idle of 6.0 so I probably want idle somewhere around 7.0-10.0 (corresponding to 1070-1100).

If the motors does not spin up at all you need to increase the Bluejay max startup power setting.
The min startup power setting is used to be able to start the motors at low throttle, e.g. maybe they need 1150 throttle to start spinning, but once they are started they can run at an idle of 1100.

Bluejay at 24kHz is supposed to have very similar startup behaviour to BLHeli_S 16.7, but I have had reports about issues with this before, so I'm interested to hear if you are still experiencing this.

Note that JazzMavericks firmware varies the throttle curve based on PWM frequency so this will usually not need one to change the power/idle settings manually, but can cause other issues with bigger motors etc.

JAL513, 2021-09-28:
Thanks for all this. Apologies for confusing things, I’m running a different target SH50 Bluejay v.13 at 96khz. It only happens when the battery gets lower and only to one motor. 

I’ve got min and max turned up a bit and definitely helps, but didn’t fix it entirely. I’ll increase idle a bit or move down to 48 or even 24khz to see if I can correct the issue even at lower battery levels.

I’ll report back after some testing.

JAL

mathiasvr, 2021-09-28:
Yes low battery power can be problematic especially on 1S, it might be a good idea to enable battery sag compensation in betaflight.

JAL513, 2021-09-29:
Increasing idle didn't result in any change. I updated to v.14 and decreased to 48kHz. I still needed to increase min and max startup almost all the way for the sputtering motor to start up with the others at low battery. After that I moved back to 96khz and kept min and max startup real high. I also moved ramp up to 10x. Flew amazing after that. Will I smoke my motors with those settings?

longxk, 2021-09-29:
v0.14 Z-H-30_48KHz target with default settings wont start up my 1202.5 motors.
after setting max startup power to 1200, the motors start to spin properly at 1130 throttle.
with DShot Idle increased to 15%, the low throttle problem seems to be solved.
the min startup power was kept at default value.
according to my experiments, min startup power seems to have no effect on my problem.

closed 2021-09-29 by longxk

mathiasvr, 2021-09-29:
@JAL513 If it's only one motor that stutters it might be more worn out or have a loose connection etc. Preferably you don't want to raise max power if other motors are fine with lower setting, but if you don't start the motors while they are stuck you should be fine. Rampup of 10x/11x is usually not a problem.

@longxk min startup have no effect if it's lower than idle. You can try setting min startup to 1150 and see if it's possible to decrease idle, but sounds like it's fine. You might also want to use `set thrust_linear = 25` in betaflight to manage some of the low throttle problems.

longxk, 2021-09-29:
@mathiasvr Max value for Min Startup Power is 1125. 😂

codeincontext, 2021-11-07:
@longxk were you able to get bidirectional DShot working on this FC? I just get 100% error rates after enabling it with ESC Configurator

longxk, 2021-11-08:
@codeincontext yes, bidirectional dshot works fine on mine

codeincontext, 2021-11-08:
> @codeincontext yes, bidirectional dshot works fine on mine

Ok, thanks