mathiasvr/bluejay #37: ESC telemetry error is not 0%

sukulent, 2021-10-05:
When testing motors in the motors tab using bidirectional dshot, the error percent isn't 0%. It scales with the throttle reaching 1% around mid throttle.

I would expected 0% error.

Is this a bug, a limitation of the L type ESCs or is this safe to fly?

**Configuration:**
 - Bluejay version: 0.14
 - ESC variant: I have two G-L-30 and two F-H-40 , only the two G-L-30 do this.
 - DShot bitrate: 300 as both L and H are supported
 - Bidirectional DShot: On
 - FC firmware: Betaflight 4.2.9

mathiasvr, 2021-10-05:
The L type ESCs have not had much testing and I don't have any myself so it's likely the telemetry timing is not tuned properly. In general I think it's preferable to use dshot 150 for L ESCs.

I could try slightly offsetting the timing if you are interested in testing?

mathiasvr, 2021-10-05:
Here are some test hex files, you can try to see if one of them improves the error percentage.

[L_Test.zip](https://github.com/mathiasvr/bluejay/files/7287847/L_Test.zip)

If both makes it worse there's probably not much more to do about it.

sukulent, 2021-10-06:
Thanks! Testing is no problem.

I read the wiki here and if I am correct I must use dshot 300 when I mix L and H type ESCs

I have flashed the two L type with both test files and here are the results:

The G_L_30_48_v0.14_predelay8.hex is better with no more than 0.1% error
The G_L_30_48_v0.14_predelay6.hex is worse with max around 1.5% error