mathiasvr/bluejay #40: Issues flashing Bluejay .14 48k to Happy Model 2G4 ELRS AIO

Zestforlife808, 2021-12-20:
I'm hoping someone can help me figure out what's going on with my new whoop board.  After confirming all motors spin in Betaflight configurator I tried flashing Bluejay .14 48k with esc-configurator.com.  

Upon powercycling the board only motor 4 spins.  The rest just make a stuttering sound without making any revolutions.  Flashing .14 24k results in motors 1 and 4 spinning while 2 and 3 stutter. 

I also tried flashing Bluejay .13 48K and only motors 1 and 4 consistently start while 2 and 3 stutter without spinning. 

I tried to revert to Blheli_S 16.7 official using the same configurator and motors 1,2, and 4 spin while 3 still stutters.  I'm fairly certain motor 3 ran without issues prior to my upgrade attempt with esc-configurator.com.  

I'm using a pre-release build of Betaflight 4.3 from this thread https://github.com/betaflight/betaflight/pull/10788 and I'm wondering if maybe the ESCs aren't flashing right since the results are all over the place.  I'd love to get Bluejay working but at this point I can't even get back to my pre-flash state.

Any help would be greatly appreciated.

mathiasvr, 2021-12-20:
Hi there, please fill out the bug report template to make it easier to diagnose issues.
Many whoops needs higher startup power so I would try that, probably min/max of 1100/1200 should be a okay but you might want to increase it in steps.

pitts-mo, 2022-01-12:
Hey @Zestforlife808, 
I have seen similar flaky motor starts happen a lot on micros. 

If you find this happens again:
Direct soldering motors to FC or squeezing the plates within the motor connector plugs on my Tinyhawk II quads allows them to start up fine again even on Bluejay's default startup power. Just be careful not to crimp the tapered end of the motor plug plates so much as to prevent them to accept the pin.  :-)
-p

malexzx, 2022-01-13:
Hey @Zestforlife808, all

Yesterday, while trying to install Bluejay to O_H_5, seen so motor 1 spin correctly, while others not.
After  investigating previous configuration, see difference: Blheli_S version is 16.8.
After load to Blheli_S 16.7 the motor 1 spin and 2,3,4 shutters.
After load to Blheli_S 16.8 all motors spin ok.

mathiasvr, 2022-01-13:
@malexzx JazzMavericks has more throttle at low end basically increasing idle. With Bluejay you need to increase min/max startup power to get the motors spinning or increase idle in the FC firmware. As @pitts-mo mentioned this could be caused by poor motor connection.

malexzx, 2022-01-13:
@mathiasvr I was already try to increase limits with no success. (Will try to direct soldering too, but it complicated)
Also while enabling bidirectional dshot I've got an error, about wrong motor number reporting by betaflight configurator.
(if it matters, I'll try to do again to report).
M.b. other  JazzMavericks things matters, can you explain?
My configuration: 
Betaflight Meteor-65 tinywhoop. 
BTFL-4.2.0 (manufacturer install)
O_H_5 16.8
target Matek-f411RX (AIO)

mathiasvr, 2022-01-13:
You might need to increase ramp up also. I would expect it to work okay with startup power min/max of about 1100/1200 but can maybe go higher, and try setting ramp up to 11x. If this doesn't work I'm not really sure how 16.8 makes it start.

malexzx, 2022-01-13:
with suggested values all motors spin ok. Thank you!

closed 2022-01-17 by mathiasvr