mathiasvr/bluejay #43: Bidirectional Dshot failing on 0.15 on Crazybee F4 PRO V3.0 - HAMO/CRAZYBEEF4FR(STM32F411)

Troyhy, 2022-01-15:
**Describe the bug**
After Enabling bidirectional Dshot-600, only motor 4 will spin, others show 100% error.
Occasionally I got all motors to spin, but could not fly, because of RMP filter warning flag.

When changed to version 0.14, worked as expected.

**Expected behavior**
All motors would show 0% error rate and no RPM filter flag preventing arm.

**Configuration:**
 - Bluejay version: 0.15
 - ESC variant: F-H-40
 - PWM frequency: 48/96 ( tested both)
 - DShot bitrate: 600
 - Bidirectional DShot: On
 - FC firmware: Betaflight 4.2.11

Troyhy, 2022-01-15:
And thanks for the Bluejay, bi-directional Dshot made Mobule7 fly really smoothly!

PMantix, 2022-01-15:
I'm having a similar issue on BetaFPV 12A AIO ELSR V1. On version 0.15, I am getting the RPM_FILTER flag and cannot arm. 

I also noticed in the motor tab that I have 100% error on all motors when the RPM is 0, but the error will drop to 0 after throttling up to any value. In version 0.14, I correctly show 0% error when RPM is 0, and do not get the RPM_FILTER flag and can arm.

kleinph, 2022-01-16:
I think you should use DSHOT600 on this board. In more recent version of Betaflight this is actually enforced.