mathiasvr/bluejay #44: v0.16 Can not be  configured

Sherhang, 2022-01-20:
**Describe the bug**

v0.16 Can not be  configured using the following configurator tools:
ESC Configurator (PWA)
Bluejay Configurator (Standalone)

I flashed v0.16 into BLHeil_S  C_H_40 successfully using Bluejay Configurator , but I can not configure it using ESC Configurator (PWA) or Bluejay Configurator。It says that version 0.16 is not supported.

**Expected behavior**

A description of what you expected to happen.

**Configuration:**

 - Bluejay version:0.16
 - ESC variant: C_H_40
 - PWM frequency: 24
 - DShot bitrate: 150 300 600
 - Bidirectional DShot: Either
 - FC firmware: Betaflight 4.2.11

mathiasvr, 2022-01-20:
Bluejay configurator is being out phased but ESC configurator should support v0.16.

Sherhang, 2022-01-21:
Thank you ! I flashed v0.16 using ESC Configurator and It works.

closed 2022-01-21 by Sherhang