mathiasvr/bluejay #47: Add the option to disable the low voltage protection, use case with liion batteries

JaKoB-FR, 2022-02-08:


mathiasvr, 2022-02-08:
The efm8 should be able to run significantly below 3V. What voltage do you want to run? Could this be a problem with voltage dropping because of on inadequate capacitor when powering the ESC?

JaKoB-FR, 2022-02-09:
> The efm8 should be able to run significantly below 3V. What voltage do you want to run? Could this be a problem with voltage dropping because of on inadequate capacitor when powering the ESC?

the esc stop starting at 5.9/5.8 V (on 2s liion which could run down to 2.5V per cells). I don't have extra capacitor on this build. This feature was common on fixed wing applications and lipo batteries when you want to have the rx and servos still running (5v) to control the glide for landing when battery is sagging. This low voltage protection can be disable on blheli32 configurator, I am wondering if it could be implemented in blue jay

mathiasvr, 2022-02-09:
There is no low voltage protection in bluejay. What ESC are you sing? I think maybe it is cutting voltage due to a regulator or similar?

iso9660, 2022-06-07:
I think that VDD supply monitor and reset source is enabled in the code below. The problem is that in order to be able to write in flash, supply monitor needs to be enabled. Maybe reset source can be disabled, but I don't know whether it could be advisable.

pgm_start:
	....
	orl	VDM0CN, #080h				; Enable the VDD monitor
	mov	RSTSRC, #06h				; Set missing clock and VDD monitor as a reset source if not 1S capable
	....