mathiasvr/bluejay #48: z_h_30 48khz?

lesharodin, 2022-02-13:
**Describe the bug**

Seems like Z_H_30 ESC's doens't work on 48khz or higher. Only 24k works.
Tested on HappyModel X12 AIO, found simmilar issues online on betafpv aio with same esc's

**Expected behavior**

Spin motors :)

**Configuration:**

 - Bluejay version: [e.g. 0.14-0.16]
 - ESC variant: [e.g. Z_H_30]
 - PWM frequency: [48/96]
 - DShot bitrate: [300]
 - Bidirectional DShot: [Either]
 - FC firmware: [e.g. Betaflight 4.3 RC3]

mathiasvr, 2022-02-13:
Increase min/max startup power to around 1100/1200.

lesharodin, 2022-02-13:
It helped, thank you very much.
Is this save to leave like this?

mathiasvr, 2022-02-13:
Your welcome! Yes I think so, the BetaFPV Z target seems to need more startup power, I haven't heard of anybody having issues after increasing it.

closed 2022-02-16 by lesharodin