mathiasvr/bluejay #49: Arming stopped after flashing v14

jamesd3rd, 2022-03-02:
I flashed my Diatone Mamba f50 BLHeli_S esc tonight and now the quad won't arm. I initially used v15 then went to v14 thinking an older version wouldn't be buggy and just work.
.
Should I be looking at  the minimum startup value?
Right now I'm using Dshot 600. Should I go down to 300?

mathiasvr, 2022-03-02:
Does it work without bidirectional dshot? What size motors are you using? PWM frequency? Does the FC not arm or is the motors just not spinning?

jamesd3rd, 2022-03-03:
There is no issue. I've resolved the problem. I had the smoke stopper plugged in and it limits the current. Once I disconnected it, things were fine. I don't normally leave it plugged in after the initial check for shorts. So I take it for granted because it's typically not plugged in when I check the motors without BF connected.

closed 2022-03-21 by mathiasvr