mathiasvr/bluejay #58: Direction of the motors does not change

astrovpls, 2022-08-31:
Describe the bug
Whichever direction of the motors I choose, they remain the same. Tried 0.16 and 0.15.
However I was able to switch direction via betaflight, with disabled bidirectional DShot.

Similar issue in #24 

Bluejay version: 0.16, 0.15
ESC variant: C-H-40
Bidirectional DShot: On
FC firmware: Betaflight 4.3.1
FC: Mamba basic f722 mini mk3
board_name: MAMBAF722_I2C

stylesuxx, 2022-08-31:
How are you changing directions? You do click "write settings" and not "flash firmware" - right?

astrovpls, 2022-09-07:
Oh, actually I was reflashing. It seemed that it was not entirely obvious from the first time when I need to reflash, and when to write down the settings.

Thanks for your reply

closed 2022-09-07 by astrovpls