mathiasvr/bluejay #59: REPO ABANDONED - NEW LOCATION

stylesuxx, 2022-08-31:
This repository is abandoned, development continues in the [new repository](https://github.com/bird-sanctuary/bluejay).

swilkens, 2022-09-08:
@mathiasvr ?

stylesuxx, 2022-09-08:
You can read more about what's going on in the [new orgs readme](https://github.com/bird-sanctuary). I am afraid you might wait for Mathias to answer.

PhilippMolitor, 2022-10-13:
you should really update the readme with a big notice to make this clear, I wasn't aware of it until i looked at the issues list.

stylesuxx, 2022-10-13:
@PhilippMolitor yeah - that's difficult to do if the repo is **ABANDONED** by the owner.

PhilippMolitor, 2022-10-13:
I wasn't aware mathias was the only one with full access, I understand :)

stylesuxx, 2022-10-13:
Unfortunately, yes - otherwise we would have loved to "take it over" properly. In the new org, 4 peoples are owners, so we should be on the safe side...