mathiasvr/bluejay #64: Can't flash in Bluejay

ThibMarot, 2023-01-19:
### Describe the issue

Hi!
I want to flash my ESC in bluejay but when i select "flash all", i can't find any bluejay ESC target.
Same problem with esc configurator and Bluejay configurator.
Can you help me?

i have the SpeedyBee F405 V3 BLS 50A 30x30 FC&ESC Stack

### Bluejay version

0.16

### ESC variant

J-H-50

### PWM frequency

24

### DShot bitrate

600

### Bidirectional DShot

Off

### FC firmware

Betaflight 4.3

Denethor85, 2023-02-13:
You have connected the battery?

ThibMarot, 2023-02-14:
yes of course

[image: Mailtrack]
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality11&>
Sender
notified by
Mailtrack
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality11&>
14/02/23
11:29:28

stylesuxx, 2023-03-01:
This repo is abandoned -> https://github.com/mathiasvr/bluejay/issues/59

I assume you have solved the issue by now, if not - come into the new discord, or post in the new repo